SHA2017-badge/micropython-esp32 #93: nvs_[get/set]_u[8/16] not working

Roosted7, 2017-07-22:
Nice to store ints in the NVS, but current code just crashes

annejan, 2017-07-22:
Can't reproduce . .

Roosted7, 2017-07-22:
Okay, if memory serves me well tsd also had this issue.
(His badge crashed before he reverted renze his commit that started using int instead of str)

annejan, 2017-07-22:
Ah, might it be that the values are stored as string, but attempt to retrieve as int? 

That would make it a legacy data issue.

Roosted7, 2017-07-22:
Ah could be.. My badge crashes when trying the get or set the voltage levels for the battery(bar)

basvs, 2017-07-23:
I also got crashes when i requested unset entries. I will dive into this

basvs, 2017-07-23:
>>> badge.nvs_get_u8("lalala", "lala", 42)
42
>>> badge.nvs_get_u8("lalala", "lala")
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.

basvs, 2017-07-23:
The code works when you use a default. If not, you will get a null-pointer-exception.

I have a patch ready.

basvs, 2017-07-23:
We cannot merge this yet. It triggers bugs in other scripts. Have to fix that first.. :)

basvs, 2017-07-23:
scripts are fixed; code is merged.

closed 2017-07-23 by basvs