SHA2017-badge/micropython-esp32 #94: Battery status on splash is completely different when awake compared to when sleeping

Roosted7, 2017-07-22:
Currently it is jumpen from 1/5 battery bar to 2/3 bar when it is going to sleep.

niekproductions, 2017-07-23:
The measured battery voltage fluctuates wildly since it is measured after the battery fuse. This means that we always measure a lower voltage than the actual battery voltage, and the difference between measured and actual voltages increases when we draw more current.
The best compensation for this unfortunate hardware design flaw is to only measure battery voltage when the badge is drawing low currents from the battery (LEDs off, no WiFi interaction and display not updating). Perhaps @rnplus can have a look if this is practical.

Roosted7, 2017-07-24:
@niekproductions The splash i'm currently working on will try to do this
- After boot
- After closing wifi
- after 100ms of idling (perhaps even lower cpu speed..)

Roosted7, 2017-07-30:
@rnplus and I have had different approaches for this problem.
At the moment it is already a lot better than the at the time of opening this issue, and will only continue to improve as we work around the hw issue is sw

closed 2017-07-30 by Roosted7