SHA2017-badge/micropython-esp32 #97: Only SSL checks for *.sha2017.org

Roosted7, 2017-07-22:
Can we include Let's encrypt?

annejan, 2017-07-22:
Yes, but how do we 'know' to use that (root) certificate?

The whole 'just *.sha2017.org' bit of code enforces certificate validation only for the domain since we know what root CA it uses . .

Roosted7, 2017-07-22:
Is it not possible to do the same for Let's encrypt? [https://letsencrypt.org/certificates/](Root certs from LE)

This kinda crypto is not my area of expertise, I'm justing doodling out what would be nice to see

basvs, 2017-07-24:
You cannot add only part of the root CAs, because you don't know which domain uses which root CA.

basvs, 2017-07-24:
I am thinking about a 'ssl-certs' egg/addon, which manages a path with tls certificates (/etc/tls/ ?)

And then could fix the tls library to automatically load root certificates from that path.

Roosted7, 2017-07-30:
Is this still a wontfix with @basvs comments?

annejan, 2017-07-30:
Well . . if there is a clear path on how to do this.
But I don't really have a nice clean idea on how to set this up (calls wise) . .

Code wise it should not be very hard, but I don't really see any way to do this and maintain compatibility with other things.

basvs, 2017-07-30:
With mbedtls_ssl_conf_verify() you can add your own callback to verify certificates.

If the library then cannot verify a certificate, this method is called. We can lookup the root certificate, then search on the filesystem for a file "/etc/certificates/".md5(cert.subject) and call the mbedtls_x509_crt_verify() method again, but now with the extra root-certificate and without the callback-handler.