SHA2017-badge/micropython-esp32 #108: missing uos.urandom(n) in emulator

basvs, 2017-07-23:


aczid, 2017-07-24:
Perhaps it's better (more portable) to use the `random` module.
I've tried to enable all its features in https://github.com/SHA2017-badge/micropython-esp32/pull/109
```
>>> import random
>>> dir(random)
['__name__', 'getrandbits', 'seed', 'randrange', 'randint', 'choice', 'random', 'uniform']
```

basvs, 2017-07-24:
The problem with urandom is that it doesn't use the ESP's hardware rng. So it's not really usable on the badge itself.

annejan, 2017-07-24:
So perhaps we should deprecate it completely?

Or not usable for crypto but useable for fun and games?

basvs, 2017-07-24:
The urandom module (and I think the random module is the same module) is a simple prng, which can be compared to srand()/rand().It's very usable if you need a (reproducible) prng-stream. It's also initialized with a constant, so if you need different random values per run, you will have to initialize it yourself.

But for stronger randomness, there's the uos.urandom() method. It's available on the badge, which uses ESP's hardware rng. But not on the emulator.

aczid, 2017-07-27:
Could you try the new port and see if this issue can be closed?

basvs, 2017-07-27:
Hmm..
    >>> uos.urandom(3)
    Segmentation fault (core dumped)

basvs, 2017-07-27:
open("/dev/urandom", O_RDONLY)          = 3
fstat(3, {st_mode=S_IFCHR|0666, st_rdev=makedev(1, 9), ...}) = 0
ioctl(3, TCGETS, 0x7ffdca3b5390)        = -1 EINVAL (Invalid argument)
read(3, "!\236ui\245\365=\233\355>\356\214\370\1\253;\7\23605.I;A\272\234J<\352(\232\235"..., 4096) = 4096
close(3)                                = 0

basvs, 2017-07-27:
fread(&vstr.buf, 1, n, fp);

there's the bug. removing the '&' will fix it

basvs, 2017-07-27:
fixed it. Issue can be closed.

closed 2017-07-27 by basvs