SHA2017-badge/micropython-esp32 #113: remove hardcoded gpio pins in main.c and modules/deepsleep.py

basvs, 2017-07-24:
Possible better solutions:

- main.c
    move the methods to components/badge/

- modules/deepsleep.py
    expose the needed pins via badge.input_pins

annejan, 2017-07-24:
I think we should always do the pin25 for touch wakeup thing probably in the mpr121 init . .

The bit in main.py is only for BPP afaik . . And they could/should be moved.

basvs, 2017-07-26:
I couldn't find any reason not to use the input framework. I've created PR #125 for that.

basvs, 2017-08-03:
The modules/deepsleep.py still contains pin 25. Not very important, but it would be nice if it uses a badge.deep_sleep() or if the deepsleep.py requests which pins are used for the input.

basvs, 2017-08-16:
Ok, found an issue with it. The V3-Lite badge doesn't have a mpr121 chip. Pin 25 is always low and immediately triggers a wake-up on deep-sleep.