SHA2017-badge/micropython-esp32 #114: Put peripherals to sleep on deepsleep

annejan, 2017-07-24:
This includes:

* LEDs
* SDcard
* E-ink

basvs, 2017-07-24:
LEDs / SDcard → we can disable the power to the LEDs/SDcard.
    (use both badge_power_leds_disable() and badge_power_sdcard_disable())

E-ink →call badge_eink_deep_sleep()

Roosted7, 2017-07-30:
@basvs Can you make those API's available to python? I think I should be able to manage that as well, but I know it costs you a fraction of that time :)

basvs, 2017-07-30:
will have a look at it

basvs, 2017-07-30:
pr #151 contains the hooks for the led and sdcard.

basvs, 2017-07-31:
pr #153 contains the hooks for the eink deep sleep

basvs, 2017-07-31:
@Roosted7 the api's are merged

sebastius, 2017-08-01:
So, is this one done?

annejan, 2017-08-01:
@Roosted7 / @rnplus are these now called on deepsleep?

Should they be added to deespleep.py ?

Roosted7, 2017-08-07:
Done
Awesome work @basvs

closed 2017-08-07 by Roosted7