SHA2017-badge/micropython-esp32 #119: Led driver not working properly above Hex 0x7F

sebastius, 2017-07-25:
badge.leds_send_data(''.join([('\x00\x00\x7F\x00') for i in range(6)]), 24)

Gives all blue LEDs at medium brightness 

badge.leds_send_data(''.join([('\x00\x00\x80\x00') for i in range(6)]), 24)

Gives white, purple, yellow, green, orange, white

badge.leds_send_data(''.join([('\x00\x00\xFF\x00') for i in range(6)]), 24)

Gives white, purple, yellow, green, orange, white but brighter.

Different techniques cause the same problem.

badge.leds_send_data('\x00\x00\xFF\x00\x00\x00\xFF\x00\x00\x00\xFF\x00\x00\x00\xFF\x00\x00\x00\xFF\x00\x00\x00\xFF\x00'), 24)

shows the same problem. Should be all leds on bright blue, instead does weird effects.

annejan, 2017-07-25:
Needs binary data, otherwise gets transformed into utf-8

Use`b'\x00\x00\xFF\x00'` etc . .

sebastius, 2017-07-25:
badge.leds_send_data(b''.join([(b'\x00\x00\xFF\x00') for i in range(6)]), 24)

works.

closed 2017-07-25 by sebastius