SHA2017-badge/micropython-esp32 #154: When waking up from touch interrupt, the information about the touch event is lost

Roosted7, 2017-07-31:
We would be able to implement very transparent sleeping in some places if we would be able to read which input triggered the interrupt. I think this is possible by not resetting the MPR121 on every python boot..?

Could we extend reset_cause() or implement something for wakeup_cause() ?

Roosted7, 2017-07-31:
Another way of implementing would be to be able set a interrupt bitmask in the MPR121, is that feasonable?

basvs, 2017-07-31:
As far as I know, you cannot find out which button triggered the wake-up. The button is already released, so the status register will not show any pressed buttons. The mpr121 also doesn't support an interrupt bitmask. It's all or nothing.

sebastius, 2017-08-01:
Impossible with this chip.

Roosted7, 2017-08-01:
Let's keep this open @sebastius.
I would love to use the ULP for this

basvs, 2017-08-01:
We have to use the I2C to access this. It might be possible, but that isn't easy to implement.

annejan, 2017-08-01:
ULP doesn't seem to be powerful enough for this from what I have been reading . .

basvs, 2017-08-03:
@Roosted7 can we close this?

closed 2017-08-03 by annejan