SHA2017-badge/micropython-esp32 #183: Add caching to installer

psy0rz, 2017-08-04:
When opening the installer and browsing through categories, the installer will re-download the lists everytime.

Implement some caching to make it faster and more reliable on slow wifi-connections.

I'll try to see if i can do it my self and send a pull request :)

psy0rz, 2017-08-04:
this is doable via a normal module, so i'm currently working on installer_cached (its in the utilities category and still work in progress)

psy0rz, 2017-08-04:
done and in beta testing now.

annejan, 2017-08-05:
So . . should we replace the base app?

And if so, can you make a PR and link this issue?

psy0rz, 2017-08-05:
nope not yet :) there is still a bug and it needs some smoothing

psy0rz, 2017-08-05:
bugs fixes, now fixing gui and trying to prevent it going back to the homescreen when an error occurs

psy0rz, 2017-08-05:
created new bugs while adding features ;)

Roosted7, 2017-08-07:
@psy0rz What is the status of your version?